Auterion/px4-ros2-interface-lib #17: Issue with example modes showing up in QGroundControl

AbhinavPeri, 2024-01-11:
Going off the documentation from the PX4 docs, I have cloned this repository and built my workspace. However when I run the examples, I am not able to find the example modes in the QGroundControl Mode Selector. I can see that the node has registered the mode and that commander status recognizes the external mode. How do I get QGroundControl to show this. Alternatively, is there another way to run an external mode without using QGroundControl?

[image]
[image]
[image]

GuillaumeLaine, 2024-01-11:
Hi @AbhinavPeri thanks for raising the issue.
This is indeed a problem, in the meantime could you try again but launching QGC _after_ running the example mode? I've also had this issue with modes not showing up properly in the ground control system if the app was already open.

AbhinavPeri, 2024-01-11:
I just tried it, and I still don't see the mode.

PerFrivik, 2024-01-11:
Hey @AbhinavPeri, try to use the QGC [daily build ](https://docs.qgroundcontrol.com/master/en/qgc-user-guide/releases/daily_builds.html) as the stable release does not have the feature implemented yet, I am guessing. 

Let me know if its works :).

AbhinavPeri, 2024-01-11:
@PerFrivik using the daily build looks like it fixed the issue. It's amazing how reinstalling software fixes so many problems. Thanks everyone for the help!

closed 2024-01-11 by AbhinavPeri

Jaeyoung-Lim, 2024-04-21:
I can confirm that this is still a problem

Kametor, 2024-10-03:
I also had a problem. If you cant open QGround Daily check this [QGroundControl Issue #9847](https://github.com/mavlink/qgroundcontrol/issues/9847).

wly218, 2025-03-22:
Hey,@azerupi.May I ask if your problem has been solved?I also couldn't diplay the custom flight mode using either of the two versions

dakejahl, 2025-03-24:
It's a bug in QGC, working on it here
https://github.com/mavlink/qgroundcontrol/pull/12581

wly218, 2025-03-25:
OK，I see.Thank you!