Auterion/px4-ros2-interface-lib #41: Executor with "activate_even_while_disarmed"

damien-robotsix, 2024-06-26:
Hello! Thank you for the library, I expect much from it.

I tried to implement an executor associated to a mode that has "activate_even_while_disarmed" setting set to true.

When you do this (with the drone being disarmed), the drone switch immediatly to the mode and does not trigger the Executor. 

While, if I have "activate_even_while_disarmed" set to false. If I switch to the mode nothing happen util I arm then the Executor works.

So two questions:
- Is it a wanted behavior? Or a bug?
- Should I have a look and propose a PR?

Best

Damien

damien-robotsix, 2024-06-26:
My objective is to have an executor that is able to arm the vehicle when I select the mode which doesn't work for now (or I didn't find out how).

bkueng, 2024-07-02:
You are correct, this case is not implemented yet. The existing "activate_even_while_disarmed" is mainly for integration tests, so they can run automatically (https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/test/integration/mode_executor.cpp#L79).
You're welcome to suggest a PR.

damien-robotsix, 2024-07-03:
Thank you for your answer.

For what I understood so far a good place to look would be 
https://github.com/Auterion/px4-ros2-interface-lib/blob/1ae6a3816fc4856f4970ea5261c1c1c90ec95c64/px4_ros2_cpp/src/components/mode_executor.cpp#L293-L294

The condition is_armed here limits the opportunity of the executor to arm itself the drone (if not activated immediately). I would add a flag in the settings that allows the executor to arm and add it to this conditions.

I guess the _is_armed condition is there for a reason (or maybe only safety?). Any side effect I should expect? Anyway I will try and see but you know better.

bkueng, 2024-07-04:
Yes you are correct, you want `wants_to_activate_immediately` true in https://github.com/Auterion/px4-ros2-interface-lib/blob/1ae6a3816fc4856f4970ea5261c1c1c90ec95c64/px4_ros2_cpp/src/components/mode_executor.cpp#L293-L294
And false here: https://github.com/Auterion/px4-ros2-interface-lib/blob/1ae6a3816fc4856f4970ea5261c1c1c90ec95c64/px4_ros2_cpp/src/components/mode_executor.cpp#L62

damien-robotsix, 2024-07-05:
Okay, I am on it, testing on my side and I will make a PR if it is working

closed 2024-07-16 by bkueng