Auterion/px4-ros2-interface-lib #50: Takeoff heading parameter has no effect

damien-robotsix, 2024-07-26:
Hello, 

Testing the take_off action (void ModeExecutorBase::takeoff). I realized that the heading was not taken into account.

Given my quick investigation, this is due to the handling on PX4 side. It seems to be completely discarded.
https://github.com/PX4/PX4-Autopilot/blob/97561d7802eddb899f1c4f0a3a3ae9aa4c26bb73/src/modules/navigator/navigator_main.cpp#L620

Or maybe I didn't get something but for now I didn't manage to set the yaw during take-off.

No big deal but maybe worth removing the variable since it has no impact.

bkueng, 2024-07-30:
Agreed, then let's remove it for now. I just thought it would be nice to have and rather make it work on the PX4 side.

JacopoPan, 2025-09-10:
Sure @bkueng , I updated https://github.com/PX4/PX4-Autopilot/pull/24040 to be an open PR

In fairness, I understand the logic of taking the heading from the mission rather than the command (it is safer: sending `MAV_CMD_NAV_VTOL_TAKEOFF` directly to the navigator requires parameters like `NAV_LOITER_RAD` and `VTO_LOITER_ALT` to be set somewhat reasonably to avoid harsh post-transition maneuvers).