Auterion/px4-ros2-interface-lib #57: How to use Local Position Updates?

XXLiu-HNU, 2024-08-30:
Hi, this work is vary useful for us to change the position  of  UAVs from GPS to local position. But when use the code we meet some errors. 

**BackGround:** 
We use RTK for UAV swarm  location. But we don't know how to use the same coordinate for each UAV.  Becase each UAV treats the starting position as the coordinate origin.  So, we want set the local_position by calculating the latitude and longitude height of the UAV.
**Methods to try:**
Fristly, we build the lib and it looks like good, and we change the value of "EKF2_EV_CTRL" from 0 to 15 and we run :
 `ros2 run example_local_navigation_cpp example_local_navigation` ,
[image]
and it also  looks like good .

But when we use : `ros2 topic echo /fmu/out/vehicle_local_position` , the locai_position is  don`t change:  
[image]

and here is the code we use, just change `px4_ros2::VelocityFrame::LocalNED` from to `px4_ros2::VelocityFrame::Unkown`, and  set the position for test, and here is the code:
```
/****************************************************************************
 * Copyright (c) 2023 PX4 Development Team.
 * SPDX-License-Identifier: BSD-3-Clause
 ****************************************************************************/
#pragma once

#include <rclcpp/rclcpp.hpp>
#include <px4_ros2/navigation/experimental/local_position_measurement_interface.hpp>

using namespace std::chrono_literals; // NOLINT

class LocalNavigationTest : public px4_ros2::LocalPositionMeasurementInterface
{
public:
  explicit LocalNavigationTest(rclcpp::Node & node)
  : LocalPositionMeasurementInterface(node, px4_ros2::PoseFrame::LocalNED,
      px4_ros2::VelocityFrame::Unknown)
  {
    _timer =
      node.create_wall_timer(1s, [this] {updateLocalPosition();});

    RCLCPP_INFO(node.get_logger(), "example_local_navigation_node running!");
  }

  void updateLocalPosition()
  {
    px4_ros2::LocalPositionMeasurement local_position_measurement {};

    local_position_measurement.timestamp_sample = _node.get_clock()->now();

    // local_position_measurement.velocity_xy = Eigen::Vector2f {1.f, 2.f};
    // local_position_measurement.velocity_xy_variance = Eigen::Vector2f {0.3f, 0.4f};

    local_position_measurement.position_z = 12.3f;
    local_position_measurement.position_xy = Eigen::Vector2f {1.f, 2.f};
    local_position_measurement.position_xy_variance = Eigen::Vector2f {0.01f, 0.01f};
    local_position_measurement.position_z_variance = 0.33F;

    local_position_measurement.attitude_quaternion = Eigen::Quaternionf {0.1, -0.2, 0.3, 0.25};
    local_position_measurement.attitude_variance = Eigen::Vector3f {0.2, 0.1, 0.05};

    try {
      update(local_position_measurement);
      RCLCPP_DEBUG(
        _node.get_logger(),
        "Successfully sent position update to navigation interface.");
    } catch (const px4_ros2::NavigationInterfaceInvalidArgument & e) {
      RCLCPP_ERROR_THROTTLE(
        _node.get_logger(),
        *_node.get_clock(), 1000, "Exception caught: %s", e.what());
    }
  }

private:
  rclcpp::TimerBase::SharedPtr _timer;
};

class ExampleLocalNavigationNode : public rclcpp::Node
{
public:
  ExampleLocalNavigationNode()
  : Node("example_local_navigation_node")
  {
    // Enable debug output
    auto ret =
      rcutils_logging_set_logger_level(get_logger().get_name(), RCUTILS_LOG_SEVERITY_DEBUG);

    if (ret != RCUTILS_RET_OK) {
      RCLCPP_ERROR(get_logger(), "Error setting severity: %s", rcutils_get_error_string().str);
      rcutils_reset_error();
    }

    _interface = std::make_unique<LocalNavigationTest>(*this);

    if (!_interface->doRegister()) {
      throw std::runtime_error("Registration failed");
    }
  }

private:
  std::unique_ptr<LocalNavigationTest> _interface;
};
```
**Desired Effect :**

when change `position_xy` and `position_z`, the UAV's local Position is  also change . Hope for useful ideas  T-T

XXLiu-HNU, 2024-08-30:
We find the `position_xy` from the topic **/fmu/in/vehicle_visual_odometry**  by read the code:[examples/cpp/navigation/local_navigation](https://github.com/Auterion/px4-ros2-interface-lib/blob/main/examples/cpp/navigation/local_navigation/include/local_navigation.hpp). By the way,   the name of the topic is should be "/xxx/out/xxxx"  for is a `Publisher`?
[image]
[image]

And how to  change the /fmu/out/vehicle_local_position is still confused.

bkueng, 2024-09-02:
I'm not quite sure what you want to achieve. If you want to have a reference position between all UAVs, why not use the global position directly?

Kametor, 2024-10-07:
I am also experiencing an issue with the ROS2 example navigation nodes. Specifically, when I run the following commands:
ros2 run example_local_navigation_cpp example_local_navigation
ros2 run example_global_navigation_cpp example_global_navigation
I do not observe any changes in the values from:
ros2 topic echo /fmu/out/vehicle_local_position
ros2 topic echo /fmu/out/vehicle_global_position
The output simply prints:
[DEBUG] [1728284976.510910435] [example_local_navigation_node]: Successfully sent position update to navigation interface.
Despite this, there is no actual change in the position values. Could you please advise what might be causing this issue, or if there are any additional steps I should take to diagnose or resolve it?
Thank you for your assistance.
[image]

bkueng, 2024-10-08:
This interface is documented here: https://docs.px4.io/main/en/ros2/px4_ros2_navigation_interface.html

closed 2024-10-11 by XXLiu-HNU

Kametor, 2024-10-18:
I have followed the instructions from the wiki and updated the EKF2_EV_CTRL parameter to 15 to receive all position and velocity updates. Afterward, I launched the example_local_navigation_node, and I received the following debug message:

[DEBUG] [1729252811.314581572] [example_local_navigation_node]: Successfully sent position update to navigation interface.

The drone is currently in Position Mode, hovering in place. However, I noticed that there is no significant change in position or velocity, despite my expectation that these values should update. Specifically, I anticipated a change in position and velocity setpoints since they were initially near zero, and I expected the drone to adjust accordingly to maintain its position.

Could you please clarify if my expectations regarding the behavior of the position and velocity setpoints are incorrect? Additionally, how can I verify that the setpoints are being sent correctly to the drone? I have also monitored the /fmu/out/vehicle_local_position topic using ros2 topic echo, and I observed that the position and velocity values do not change as expected.

Any guidance on troubleshooting or validating the setpoint updates would be appreciated.

Kametor, 2024-10-18:
I have a theory regarding the issue I'm encountering. I am providing position and velocity updates, but it seems that the EKF (Extended Kalman Filter) is comparing these updates with data from other sensors, such as the accelerometer and GPS. Since the EKF may determine that my updates significantly differ from the sensor data, it may reject or downweight these updates during the sensor fusion process. As a result, my setpoints become ineffective because the drone is stationary, and no meaningful movement occurs.

If this theory is correct, it raises concerns about the practicality of this example for demonstration purposes. The tutorial may not be functioning as intended because the position and velocity updates aren't being used effectively in scenarios where the drone remains in hover.

Could you confirm if this is the case? If so, it would suggest that the current example might not be suitable for showcasing this functionality in a static or near-static situation.

Kametor, 2024-12-02:
Thanks for the clarification. I'd be happy to create a PR to update the code/docs and provide more clarity on this. I'll get started on it. Let me know if you have any specific suggestions or if there's anything else you'd like included.

Percylevent, 2024-12-17:
> Thanks for the clarification. I'd be happy to create a PR to update the code/docs and provide more clarity on this. I'll get started on it. Let me know if you have any specific suggestions or if there's anything else you'd like included.

I am in great hurry，I want to know if your theory  is correct？And how you fix it？