Auterion/px4-ros2-interface-lib #58: Compilation error in example 'example_mode_manual'

Jimi1811, 2024-09-26:
## Environment
I am using the Holybro Pixhawk Jetson Baseboard (Jetson Orin NX with ROS 2 Humble + Pixhawk V6X with PX4 1.15). 

## Description
I am compiling the main branch of this repository to execute the example 'example_mode_manual'. With this I am using the package `px4_msgs` with the main branch (both packages compiled correctly). I have this result:

```bash
$ ros2 run example_mode_manual_cpp example_mode_manual
[DEBUG] [1727366736.217726203] [example_mode_manual]: Waiting for FMU...
[DEBUG] [1727366736.499613228] [example_mode_manual]: Checking message compatibility...
[DEBUG] [1727366736.508965524] [example_mode_manual]: Subscriber found, continuing
[DEBUG] [1727366736.509311580] [example_mode_manual]: Publisher found, continuing
[ERROR] [1727366736.831281315] [example_mode_manual]: Mismatch for the following topics, update PX4 or the px4_ros2 library and px4_msgs:
  - fmu/in/aux_global_position
  - fmu/in/vehicle_attitude_setpoint
  - fmu/out/vehicle_global_position
  - fmu/out/vehicle_local_position
terminate called after throwing an instance of 'std::runtime_error'
  what():  Registration failed
[ros2run]: Aborted

```

I tried to change the repository to the tag 1.2.7, but when the package is compiling I get this result:
``` bash
$ colcon build --base-paths ./src/px4-ros2-interface-lib
Starting >>> px4_ros2_cpp
[Processing: px4_ros2_cpp]                             
--- stderr: px4_ros2_cpp                               
/workspaces/isaac_ros-dev/src/WHOX_Autonomy_stack/px4-ros2-interface-lib/px4_ros2_cpp/src/control/setpoint_types/experimental/attitude.cpp: In member function ‘void px4_ros2::AttitudeSetpointType::update(float, float, float, const Vector3f&, float)’:
/workspaces/isaac_ros-dev/src/WHOX_Autonomy_stack/px4-ros2-interface-lib/px4_ros2_cpp/src/control/setpoint_types/experimental/attitude.cpp:53:6: error: ‘using VehicleAttitudeSetpoint = struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’ {aka ‘struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’} has no member named ‘roll_body’
   53 |   sp.roll_body = roll;
      |      ^~~~~~~~~
/workspaces/isaac_ros-dev/src/WHOX_Autonomy_stack/px4-ros2-interface-lib/px4_ros2_cpp/src/control/setpoint_types/experimental/attitude.cpp:54:6: error: ‘using VehicleAttitudeSetpoint = struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’ {aka ‘struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’} has no member named ‘pitch_body’
   54 |   sp.pitch_body = pitch;
      |      ^~~~~~~~~~
/workspaces/isaac_ros-dev/src/WHOX_Autonomy_stack/px4-ros2-interface-lib/px4_ros2_cpp/src/control/setpoint_types/experimental/attitude.cpp:55:6: error: ‘using VehicleAttitudeSetpoint = struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’ {aka ‘struct px4_msgs::msg::VehicleAttitudeSetpoint_<std::allocator<void> >’} has no member named ‘yaw_body’
   55 |   sp.yaw_body = yaw;
      |      ^~~~~~~~
gmake[2]: *** [CMakeFiles/px4_ros2_cpp.dir/build.make:230: CMakeFiles/px4_ros2_cpp.dir/src/control/setpoint_types/experimental/attitude.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
gmake[1]: *** [CMakeFiles/Makefile2:163: CMakeFiles/px4_ros2_cpp.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_ros2_cpp [33.3s, exited with code 2]

Summary: 0 packages finished [33.7s]
  1 package failed: px4_ros2_cpp
  1 package had stderr output: px4_ros2_cpp
  8 packages not processed

```

mrpollo, 2024-09-26:
@bkueng any ideas? maybe related to msgs?

Jimi1811, 2024-09-26:
It starts working with the `release/1.15` branch. This branch also works with the `main` branch and the `1.2.7` tag.

dakejahl, 2024-09-26:
@Jimi1811 did you see this in the README? An easy fix is to just copy over the msgs/ directory from PX4 into px4_msgs/ package

> To manually verify that two local versions of PX4 and px4_msgs have matching message sets, you can use the following script:
>> ./scripts/check-message-compatibility.py -v path/to/px4_msgs/ path/to/PX4-Autopilot/

Jimi1811, 2024-09-26:
Actually I didn't, I just followed the documentation from PX4, rookie mistake. Thanks!

closed 2024-09-26 by Jimi1811