Auterion/px4-ros2-interface-lib #60: Clarification on Manual Mode Behavior and Expected Output

Kametor, 2024-10-07:
Hi,
I have attempted running the manual mode example, and from examining the code, it appears that this mode primarily increments the manual input values. Would it be correct to assume that the intended functionality is just to increment these values?
I am considering whether it might be more effective to set specific manual values directly, as that could make the resulting changes easier to observe.
Additionally, I am uncertain if my system is behaving as expected, so I would appreciate some clarification on the expected behavior for this example. Could you provide some guidance on what I should be seeing when the system is working correctly?
Thank you for your assistance.

bkueng, 2024-10-08:
Which part do you refer to specifically? The [heading](https://github.com/Auterion/px4-ros2-interface-lib/blob/main/examples/cpp/modes/manual/include/mode.hpp#L55)?
The example mode implements [Rattitude](https://docs.px4.io/v1.11/en/flight_modes/rattitude_mc.html).

Kametor, 2024-10-10:
Actually I mean the attitude and rate. But I see the purpose now. Thank you.

closed 2024-10-10 by Kametor