Auterion/px4-ros2-interface-lib #61: Issue Running RTL Example After Build and Source

Kametor, 2024-10-17:
Hi,
I have encountered an issue while trying to run the RTL example from the example_mode_rtl_replacement_cpp package in my ROS 2 workspace.

After successfully building the workspace and sourcing the environment, I attempt to run the example with the following command:
bash
ros2 run example_mode_rtl_replacement_cpp example_mode_rtl_replacement
However, this results in the following error:

yaml
No executable found
I've checked the build process, and there are no obvious errors. I also verified that I sourced the workspace correctly. Despite this, the executable doesn't seem to be recognized when I attempt to run it.

Could you please assist me in identifying the issue, or if possible, could you try running this example on your end to see if the same problem occurs? Any insights or suggestions you can provide would be greatly appreciated.

Thank you for your support.

GuillaumeLaine, 2024-10-17:
Hi @Kametor,
It looks like you've got [the wrong executable](https://github.com/Auterion/px4-ros2-interface-lib/blob/7b0162d394cc94dbc8214277f1c74733a27f112e/examples/cpp/modes/rtl_replacement/CMakeLists.txt#L21). Can you try this run command instead?

```
ros2 run example_mode_rtl_replacement_cpp example_mode_rtl
```

Kametor, 2024-10-17:
It worked, thanks

closed 2024-10-17 by GuillaumeLaine