Auterion/px4-ros2-interface-lib #64: Add Python support

damanikjosh, 2024-11-30:
I’ve reviewed this library and noticed that its code seems adaptable to Python. Since Python is widely used for quick prototyping and scripting, adding Python support could make the library more accessible and beneficial for a larger group of researchers and enthusiasts.

Would the team be open to exploring this? I’d be happy to discuss ideas or assist with the translation.

Thank you for your great work on this project!

bkueng, 2024-12-02:
Yes Python is definitely on our list. Great if you want to look into that!
We can discuss here, or you can ping me on [PX4 discord](https://docs.px4.io/main/en/contribute/support.html#forums-and-chat) if you want to discuss more directly.

Jaeyoung-Lim, 2025-12-23:
+1 for adding python support.

@bkueng I tried using nanobind to bind python to the interface lib, but seems like it would still be running rclcpp.
Would you have any insights on how this could be done?

bkueng, 2026-01-06:
Exactly, that's the implication. If you want other subscriptions in python, you'll effectively need 2 nodes.
It's not nice, but I think it would still be usable.

The alternative would be to completely implement the library in python. Which would be more effort, also to maintain both versions.