Auterion/px4-ros2-interface-lib #66: Custom ROS modes not visible in QGroundcontrol for Pixhawk 6X

NilsRublein, 2024-12-13:
# Issue Summary
We are experiencing an issue using the PX4 ROS2 Interface Library with our custom ROS2 package (`px4_ros2_trajectory_control`, a slightly modified version of your [experimental trajectory controller](https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/src/control/setpoint_types/experimental/trajectory.cpp)) to control our custom drone (Sarax). While everything works as expected in simulation and with Pixhawk 5X hardware, the ROS Trajectory Mode does not appear in QGroundControl when using Pixhawk 6X, despite identical software versions, workflow and setup.

---

# Context
- **Goal:** Use the PX4 ROS2 Interface Library to send velocity (trajectory) commands via ROS2 for safe motion control of Sarax (Pixhawk 6X).
- **Setup:** 
  - Ground station PC: Running QGroundControl.
  - Sarax: Equipped with a Jetson Pixhawk Baseboard (Pixhawk 6X).
  - Communication: Telemetry radio between the ground station and Pixhawk 6X.
  - ROS2 node: Runs directly on the Jetson board, translating Twist commands to PX4-compatible setpoints.

---

# Observations
1. **Simulation (X500 in Gazebo):**  
   ROS Trajectory Mode appears in QGroundControl and functions correctly.  
   - PX4 version: v1.15.0  
   - Communication: Direct UDP  
   - Hardware: Simulation only  
2. **Jetson with a separate Pixhawk 5X:**  
   ROS Trajectory Mode does appear in QGroundControl.  
   - PX4 version: v1.15.0
   - Communication: Telemetry radio or UDP  
   - Hardware: Pixhawk 5X  
3. **Jetson with build-in Pixhawk 6X:**  
   ROS Trajectory Mode does not appear in QGroundControl.  
   - PX4 version: v1.15.0 
   - Communication: Telemetry radio or UDP 
   - Hardware: Pixhawk 6X  

---

# Findings
1. The issue seems to be hardware-dependent:  
   - **Works:** Pixhawk 5X (via telemetry radio or direct connection).  
   - **Fails:** Pixhawk 6X (even with identical firmware and setup, also tested for firmware versions v1.15.1 and v1.15.2).  
2. ROS Trajectory Mode is initialized correctly but never activated because it does not become visible in QGroundControl.  
3. MAVLink console outputs for both hardware versions show no significant differences.
4. Flashing the firmware via QGroundControl or building from source made no difference.
5. Using telemtry or UDP for communication does also not seem to make a difference.
6. We also tried using separate PCs (Ubuntu 22.04) to see whether this is a Jetson issue, which yielded the same results.  

---

# Request
Given that the issue seems tied to Pixhawk 6X hardware, could there be a firmware or PX4 ROS2 Interface Library compatibility issue affecting custom modes over telemetry? Is there a different way of activating the ROS mode other than from QGroundControl? We would appreciate any insights or suggestions for debugging this further.

mrpollo, 2024-12-13:
Can you please clarify how are the Pixhawk's connected to the Jetson? (eg: ethernet, serial)

NilsRublein, 2024-12-16:
Good question! 

I made some mistakes in the list above, let me clarify:
* In the list above, communication (telemetry radio (serial) or Ethernet (UDP)) referred to the connection between PX4 and QGroundControl (QGC).
* Observation 2. and 3. were done with a separate PC (Ubuntu 22.04) serving as onboard computer, not the Jetson Pixhawk board. Here, the PC was connected via Ethernet (UDP) to the Pixhawk's. 
  * Here it was checked if the custom ROS mode would also show up on another PC on QGC over the telemetry radio (serial), which it did.
* The Jetson and the build-in Pixhawk are connected via an Ethernet switch as shown [here](https://holybro.com/products/pixhawk-jetson-baseboard?srsltid=AfmBOoog5XpaIPZ-oj_SCGsaPHXPD2AH8xP6Jlv-UduDq2cDbQZrImSs). 
  * Here we only tested with telemetry radio as communication method for QGC as it wouldn't make sense to run QGC on the Jetson itself. However, from Observation 2. and 3. we can infer that the ROS mode not showing up is dependent on the Pixhawk that is being used and is independent from the Jetson.

Here's the corrected list:

# Observations
1. **Simulation (X500 in Gazebo):**  
   ROS Trajectory Mode appears in QGroundControl and functions correctly.  
   - PX4 version: v1.15.0  
   - QGC Communication: Ethernet (UDP)  
   - Hardware: Simulation only  
2. **PC with a separate Pixhawk 5X:**  
   ROS Trajectory Mode does appear in QGroundControl.  
   - PX4 version: v1.15.0
   - QGC Communication: Telemetry radio (serial) or Ethernet (UDP)    
   - Hardware: Pixhawk 5X 
3. **PC with a separate Pixhawk 6X:**  
   ROS Trajectory Mode does not appear in QGroundControl.  
   - PX4 version: v1.15.0
   - QGC Communication: Telemetry radio (serial) or Ethernet (UDP)  
   - Hardware: Pixhawk 6X 
4. **Jetson with build-in Pixhawk 6X:**  
   ROS Trajectory Mode does not appear in QGroundControl.  
   - PX4 version: v1.15.0 
   - QGC Communication: Telemetry radio (serial)
   - Hardware: Pixhawk 6X

bkueng, 2024-12-17:
Thanks for the detailed report @NilsRublein. The mavlink messages for the modes are still under development and were thus not enabled on all boards. You need to add [this line](https://github.com/PX4/PX4-Autopilot/blob/v1.15.1/boards/px4/fmu-v5x/default.px4board#L75) to the v6x board: https://github.com/PX4/PX4-Autopilot/blob/v1.15.1/boards/px4/fmu-v6x/default.px4board.
On `main` this is already the case now.

NilsRublein, 2024-12-18:
Thank you @bkueng, this worked!

NilsRublein, 2024-12-18:
Closing this issue as solved.