Auterion/px4-ros2-interface-lib #78: Why Trajectory settpoint is split into two parts?

benchaso, 2025-02-13:
The trajectory setpoint is split into two part "update" and "updatePosition"?

This make not possible to send position and yaw at the same time. So if wanted to do yaw tracking by keeping the same position is not possible?

bkueng, 2025-02-17:
Because position and velocity should not be used together generally to prevent inconsistencies.
It sounds like you better use the [goto setpoint](https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/include/px4_ros2/control/setpoint_types/goto.hpp) for your use-case.

benchaso, 2025-02-17:
Ok, thanks. In fact, goto setpoint could be better

closed 2025-02-17 by benchaso

azerupi, 2025-02-17:
>Because position and velocity should not be used together generally to prevent inconsistencies.

Does this apply if we control individual axis? For example control x, y in velocity but z in position.

Isn't altitude mode in PX4 implemented with position + acceleration setpoints? The current API doesn't seem like it would allow us to implement our own altitude mode. Is that correct?

azerupi, 2025-02-19:
@bkueng thanks for the clarification!

>And you can also define your own setpoint type class (e.g. by copying TrajectorySetpointType) in your code and then send any of the PX4 setpoint msg types.

That's a good point, I was fixated in the API of the library that I forgot we could subclass the type and extend it however we want. If this is a commonly expected pattern it might be a good idea to call it out in the PX4 docs on external modes?

>Generally, yes PX4 is making use of that, but also makes sure it's not sending inconsistent setpoints. I'm fine with exposing the full set, but it needs a clear 'you need to know what you are doing' kind of comment.

If I were to send a PR in that direction, how would you prefer the interface to look like?

1. Overload the `update` method to take in all parameters (position, velocity, acceleration, yaw, yaw rate)
2. Overload the `update` method to take the px4 message type directly
3. Define a new struct in the same way as `LocalPositionMeasurement` where all fields are wrapped in `std::optional` that is then passed to an overloaded `update` method? Anything that isn't set explicitly it set to `NaN`. Would we want to separate all axis into their own optional so that one can set position only on one axis without the risk of setting the other two axis to 0 instead of `Nan`
4. Something else?

bkueng, 2025-02-20:
> If this is a commonly expected pattern it might be a good idea to call it out in the PX4 docs on external modes?

This section already mentions it: https://docs.px4.io/main/en/ros2/px4_ros2_control_interface.html#setpoint-types.
I think it won't be a very common pattern, as the common use-cases should be covered by the library already (at least longer-term).
But maybe we can make it more explicit, if you missed that?

> If I were to send a PR in that direction, how would you prefer the interface to look like?

I would go with 3, as it's becoming too complex otherwise. I'm not sure if we should distinguish between horizontal vs vertical or all 3 axis individually. If we use the builder pattern we can go with 3 individual axis and add helper methods in the form of `SetpointStruct& withHorizontalPosition(const Eigen::Vector2f& position);` to make it easy to use.