Auterion/px4-ros2-interface-lib #102: Custom mode not showing in QGroundControl

bentwookie, 2025-04-16:
I believe I'm following the instructions [here](https://docs.px4.io/main/en/ros2/px4_ros2_control_interface.html) and while much of it seems to be working, I don't see My Manual Control in QGC. 

I'm using main PX4 (git clone https://github.com/PX4/PX4-Autopilot.git --recursive)
I've verified with the check-message-compatibility.py script that my px4_msgs are compatible
When following the instructions I do see the ROS node emit:

```[DEBUG] [1744835601.210782361] [example_mode_manual]: Waiting for FMU...
[DEBUG] [1744835601.491691253] [example_mode_manual]: Checking message compatibility...
[DEBUG] [1744835601.495339180] [example_mode_manual]: Subscriber found, continuing
[DEBUG] [1744835601.495465581] [example_mode_manual]: Publisher found, continuing
[DEBUG] [1744835601.782706427] [example_mode_manual]: Registering 'My Manual Mode' (arming check: 1, mode: 1, mode executor: 0)
[DEBUG] [1744835601.782828295] [example_mode_manual]: Subscriber found, continuing
[DEBUG] [1744835601.783005537] [example_mode_manual]: Publisher found, continuing
[DEBUG] [1744835601.793667430] [example_mode_manual]: Got RegisterExtComponentReply
[DEBUG] [1744835601.806769636] [example_mode_manual]: Arming check request (id=1, only printed once```

and px4's console shows My Manual Mode in `commander status`:

```INFO  [commander] Disarmed
INFO  [commander] navigation mode: Hold
INFO  [commander] user intended navigation mode: Hold
INFO  [commander] in failsafe: no
INFO  [commander] External Mode 1: nav_state: 23, name: My Manual Mode
commander: cycle: 41204 events, 8000us elapsed, 0.19us avg, min 0us max 4000us 27.864us rms
commander: preflight check: 4579 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms```

However, in the daily QGC (37575aa 64bit), I don't see the mode change, nor do I see My Manual Mode in the list. I can take off and see the x500 fly in Gazebo. 

I'm a little new to this. Is something broken or is there something about my setup?

dakejahl, 2025-04-16:
@bentwookie  You might need to restart QGC, check out the discussion here
https://github.com/mavlink/qgroundcontrol/pull/12581#issuecomment-2748305828

bentwookie, 2025-04-17:
Ah. That worked. I'm able to see My Manual Mode as well as run Go-to Example.

closed 2025-04-17 by bentwookie