Auterion/px4-ros2-interface-lib #108: Support sending GPS_GLOBAL_ORIGIN and DO_SET_HOME commands

tedlin0913, 2025-05-13:
Hello!
Thank you for this great work. My goal is to use this library together with VIO for position estimation in order to fly my drone in GNSS-denied areas. To achieve this, I followed the px4 guide on [enable auto mode with local position](https://docs.px4.io/main/en/ros/external_position_estimation.html#enabling-auto-modes-with-a-local-position), which suggests sending the `VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN` command. In addition, `VEHICLE_CMD_DO_SET_HOME` can be used to set the home position once the global position estimate is available.
I am now trying to figure out how to send these two VehicleCommands correctly using this library. So far, I have attempted to send the commands via `sendCommandSync` in my mode executor `ModeExecutorTest`, which inherits from `ModeExecutorBase`. The functions I written for sending the command looks like this:

```c++
void ModeExecutorTest::setHome(const CompletedCallback &on_completed)
{
    if (is_home_set_)
    {
        on_completed(Result::Success);
        return;
    }
    // TODO: check if global position is available before sending the command? 
    const Result result = sendCommandSync(
        px4_msgs::msg::VehicleCommand::VEHICLE_CMD_DO_SET_HOME,
        1.f);

    // TODO: maybe subscribe HomePosition.msg to check if home position is availible?

    if (result != Result::Success)
    {
        on_completed(result);
        return;
    }
}

void ModeExecutorTest::setGlobalOrigin(const CompletedCallback &on_completed)
{
    if (is_origin_set_)
    {
        on_completed(Result::Success);
        return;
    }
   
    // VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN does not send back a VehicleCommandAck.
    sendCommandSync(
        px4_msgs::msg::VehicleCommand::VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN,
        NAN, NAN, NAN, NAN,
        0.0, 0.0, 0.0);
}
```
VEHICLE_CMD_DO_SET_HOME sends back a VehicleCommandAck, whereas VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN does not.

I plan to use these commands in `runState`. Two follow up questions occur: 
1. How can I check, inside the mode executor, when the global position estimate becomes available, before setting the home position?
2. How can I subscribe to `HomePosition.msg` to detect when the home position becomes available, and then set `on_completed` to `Result::Success` using this library?

bkueng, 2025-05-14:
Hi
Glad to see you got it to work.
For the global position there's already a class here: https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/include/px4_ros2/odometry/global_position.hpp. And the home position is here: https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/include/px4_ros2/vehicle_state/home_position.hpp.

We should also check why `VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN` does not return an ack on the PX4 side.

You're welcome to create a PR, but I'd rather not add them to `ModeExecutorBase` to avoid bloating the base class. Maybe something like a `HomePositionSetter` class, which takes a `ModeExecutorBase` reference to send the command.

tedlin0913, 2025-05-15:
@bkueng Thank you for the comment! 

> Glad to see you got it to work.
For the global position there's already a class here: https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/include/px4_ros2/odometry/global_position.hpp. And the home position is here: https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/include/px4_ros2/vehicle_state/home_position.hpp.

I am aware that there are existing classes, and I agree that using them would be more consistent. 
From the examples, I noticed that shared pointers to these classes are usually created inside the constructor of a class derived from `ModeBase`. Since I am using `sendCommandSync`, maybe I should pass in a reference of a custom mode, such as using `ownedMode()` in this case?

https://github.com/Auterion/px4-ros2-interface-lib/blob/f619f50a3540102f4d5653101ffe909ab7061f2a/px4_ros2_cpp/include/px4_ros2/odometry/global_position.hpp#L25

> We should also check why VEHICLE_CMD_SET_GPS_GLOBAL_ORIGIN does not return an ack on the PX4 side.

After some investigation, I discovered that this feature wasn’t included in v1.15.4 (the version I am using), but it has been fixed in the main branch. I am going to manually update the firmware for now.

tag v1.15.4
https://github.com/PX4/PX4-Autopilot/blob/99c40407ffd7ac184e2d7b4b293f36f10fe561ef/src/modules/ekf2/EKF2.cpp#L465C1-L483C1

main
https://github.com/PX4/PX4-Autopilot/blob/cadf69f73c2212e409ab9f8228a077eef914cadf/src/modules/ekf2/EKF2.cpp#L502C1-L532C1

> You're welcome to create a PR, but I'd rather not add them to ModeExecutorBase to avoid bloating the base class. Maybe something like a HomePositionSetter class, which takes a ModeExecutorBase reference to send the command.

A Setter class sounds like a great solution—perhaps something like a SetterBase, dedicated to sending commands and receiving command ack?

bkueng, 2025-05-16:
>  Since I am using sendCommandSync, maybe I should pass in a reference of a custom mode, such as using ownedMode() in this case?

Correct.
Or you could add them as a member to your mode and then access that from the ModeExecutor through the owned mode. This is slightly more cumbersome though.

> A Setter class sounds like a great solution—perhaps something like a SetterBase, dedicated to sending commands and receiving command ack?

Yes, sounds good.

bkueng, 2025-07-23:
@tedlin0913 were you able to do the changes?

bkueng, 2025-07-30:
Sounds good, we can keep it open until then.