Auterion/px4-ros2-interface-lib #127: Manual control example is not working

patrikpordi, 2025-06-30:
I have tried to create my own manual control, based on the example given, however the drone sinks to the ground no matter what. I have tried to debug it in my code, checked all the values published at `fmu/out/manual_control_setpoint` and it seemed good, therefore I suspected the controller being broken. To verify I also run the example given, the drone was sinking to the ground regardless the throttle value.
I am using PX4 main.

bkueng, 2025-07-07:
Messages are now versioned for compatibility. You have to update px4_msgs to the v1 version of the topic: https://github.com/PX4/px4_msgs/blob/main/msg/VehicleAttitudeSetpoint.msg#L1

patrikpordi, 2025-07-23:
The example did not work for me properly but I was able to implement my own based on it that worked!
https://github.com/ARK-Electronics/ros2_px4_teleop_example
Thanks!

closed 2025-07-23 by patrikpordi