Auterion/px4-ros2-interface-lib #128: Switching to External 1 is currently not possible.

migdf01, 2025-07-04:
I'm trying to try the manual example from the repository and it works well for the most part. Nonetheless, sometimes Pixhawk can't switch to the "My Manual mode", prompting the following warning in QGC: _Switching to External 1 is currently not possible_

[image]

I follow the steps as written in the guide of the repository (arming and then switching to the custom mode), but yet again, sometimes this error occurs and I don't know if there is a parameter in QGC that can be changed or a .hpp that needs to be edited to override this situation. If you have any solutions, please let me know!

**FMU:** Pixhawk 6x
**Vehicle:** Fixed wing
**OS** Ubuntu 22.04
**Equipment:** Raspberry Pi 4

Thank you!

bkueng, 2025-07-07:
Is the RC signal intermittent?
You should see it when switching to the mode before arming in the QGC status (it turns red -> clicking on it shows the error).

migdf01, 2025-07-10:
> I have not experienced that myself, so I cannot help much. Is this with latest PX4?

I'm using PX4 version 1.16.0

> The defaults should generally be fine, unless you're hitting CPU load limits. For fixed-wing you can reduce the rate of direct actuator setpoints to around 50-100Hz.

I reduced the rate of direct actuator rates to 50 Hz changing this line in direct_actuators.hpp but I'm not really sure if that makes any effect.

[image]