Auterion/px4-ros2-interface-lib #141: Release tag 1.16.0 and 1.16.1 are not compatible with PX4 1.16.0

sansha, 2025-08-19:
In PX4 main (and therefore also px4_msgs main) we have this file: https://github.com/PX4/PX4-Autopilot/blob/main/msg/versioned/FixedWingLateralSetpoint.msg
that is used by https://github.com/Auterion/px4-ros2-interface-lib/blob/main/px4_ros2_cpp/src/control/setpoint_types/fixedwing_lateral_longitudinal.cpp

The message does, however not exist in the PX4 `release/1.16` branch and also not in the now released `1.16.0` tag.
Therefore using the px4 ros2 interface library with this PX4 version in `main`, `1.16.0` and `1.16.1` leads to a build error

```
--- stderr: px4_ros2_cpp                               
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::LateralControlConfiguration_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::LongitudinalControlConfiguration_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::FixedWingLongitudinalSetpoint_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::FixedWingLateralSetpoint_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/px4_ros2_cpp_unit_tests.dir/build.make:275: px4_ros2_cpp_unit_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:273: CMakeFiles/px4_ros2_cpp_unit_tests.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::LateralControlConfiguration_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::LongitudinalControlConfiguration_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::FixedWingLongitudinalSetpoint_<std::allocator<void> > >()'
/usr/bin/ld: libpx4_ros2_cpp.so: undefined reference to `rosidl_message_type_support_t const* rosidl_typesupport_cpp::get_message_type_support_handle<px4_msgs::msg::FixedWingLateralSetpoint_<std::allocator<void> > >()'
collect2: error: ld returned 1 exit status
gmake[2]: *** [CMakeFiles/integration_tests.dir/build.make:243: integration_tests] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:218: CMakeFiles/integration_tests.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< px4_ros2_cpp [1.61s, exited with code 2]
```

Using the `release/1.16` branch fixes this, but it seems like this change was not yet released.

closed 2025-08-20 by bkueng

sansha, 2025-08-20:
@bkueng thank you for the clarification! We'll be using the `release/1.16` branch then. 
How do you plan on dealing with point releases of the 1.16 release? It would be great if there was a stable way of ensuring compatibility with each point release.

What do the version tags represent in this repo?

Thank you, understanding this better helps a lot!

bkueng, 2025-08-21:
I expect the point releases won't add breaking message changes, as PX4 generally only adds bug fixes to release branches.
Should that assumption turn out to be invalid, we can still integrate a more fine-grained approach.

The versions are used to build debian binaries, which we use internally. We may have to split that out at some point as it is indeed a bit confusing.