Auterion/px4-ros2-interface-lib #150: Adding Loiter Mode to executor with multiple modes

RoyCRO, 2025-09-24:
Hi, 

I would like to know if it is possible to add loiter/hold mode after my custom modes have completed.  

Using: release/1.16 branch

I tried to add a loiter case state as shown below, but I am unable to build it successfully. 

```
enum class State
  {
    Reset,
    TakingOff,
    MyFirstMode,
    MySecondMode,
    MyThirdMode,
    Loiter, // want to add loiter/hold mode here.
    RTL,
    WaitUntilDisarmed,
  };
```

```
case State::MyThirdMode:
        scheduleMode(
          _third_mode.id(), [this](px4_ros2::Result result) {
            runState(State::Loiter, result);
          });
        break;

case State::Loiter:
        loiter([this](px4_ros2::Result result) {runState(State::RTL, result);});
        break;
```

Error Message:

> bin/ld: CMakeFiles/search_mode.dir/src/main.cpp.o: in function `DrawModeExecutor::runState(DrawModeExecutor::State, px4_ros2::Result)':
> main.cpp:(.text._ZN16DrawModeExecutor8runStateENS_5StateEN8px4_ros26ResultE[_ZN16DrawModeExecutor8runStateENS_5StateEN8px4_ros26ResultE]+0x502): undefined reference to `px4_ros2::ModeExecutorBase::loiter(std::function<void (px4_ros2::Result)> const&)'
> collect2: error: ld returned 1 exit status
> gmake[2]: *** [CMakeFiles/search_mode.dir/build.make:160: search_mode] Error 1
> gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/search_mode.dir/all] Error 2
> gmake: *** [Makefile:146: all] Error 2

bkueng, 2025-09-25:
Hi

This functionality does not exist in `ModeExecutorBase`. You can implement it with the existing `scheduleMode` and a timer, like this:
```
scheduleMode(ModeBase::kModeIDLoiter, []{});
_timer = _node.create_wall_timer(4s, [this] {
            runState(State::RTL, px4_ros2::Result::Success);
});
```

aton-of-data, 2026-09-12:
The `scheduleMode(ModeBase::kModeIDLoiter, ...)` workaround still applies on `release/1.17`
(`4a3370f`). One caveat worth knowing if Loiter is used as a resting state in the executor.

`kModeIDLoiter` is `NAVIGATION_STATE_AUTO_LOITER` (4), the nav_state PX4 enters when it
switches to Hold (`ModeManagement.cpp:358-361`), so this shows up as Hold in QGC rather than
as a separate mode. Scheduling it keeps the executor in charge, so the state machine survives
the transition.

The caveat is that while a scheduled internal mode is active, the executor process is no
longer covered by any PX4 liveness check. Commander's external-component check acts only on a
registration whose `nav_mode_id` is set, so with `AUTO_LOITER` active a dead companion process
is invisible to it. Measured on PX4 v1.17.0 SITL (SIH, headless, one run, recorded config):
with the executor in scheduled Hold, `SIGKILL` on the process left the vehicle in
`AUTO_LOITER` for the rest of the run, with no unresponsive event and no failsafe. The same
`SIGKILL` with the owned mode active produced `AUTO_RTL` within 1.268 s (ULog, upper bound).
Run configs and metrics for both are at
https://github.com/aton-of-data/guara-llm-embodied-flight/tree/main/docs/evidence/20260911T222950Z_kill_arbiter_in_hold_s42
and `20260911T222916Z_kill_arbiter_in_cf_s42` beside it.

So the bound on how long the aircraft may sit in Hold has to come from the application. The
wall timer in the snippet above is the mechanism that provides it. Keeping it short and
re-arming it from the executor means a process death during Hold ends in `rtl()` instead of an
indefinite hover.